SasaKaranovic/OpenFanController #1: Scale for more fans

Miggas, 2023-12-17:


Miggas, 2023-12-17:
Also would it be possible to implement the ability to use temperature sensors?, I'm currently using 2, and this would be useful to make a Fan/Pump curve to the water temperature instead of the CPU/GPU temps

SasaKaranovic, 2024-01-28:
Hey @Miggas 

Yes it would be fairly easy to add support for one or more temperature sensors.
Are you using them with OpenFAN board? How many sensors do you have and what type/pn?

Miggas, 2024-01-28:
Temperature probe type: NTC, 10KΩ at 25°C
They are the most common type of temp probe used in water cooling loops.

SasaKaranovic, 2024-01-29:
Currently OpenFAN does not support analogue sensors, only digital ones (I2C or 1-Wire).
But maybe in the next revision we can add support for analogue/NTC sensors.

closed 2024-01-29 by SasaKaranovic

lietu, 2024-12-04:
Why is this closed? I don't see any support for NTC thermal probes which are incredibly standard in the PC world.

laleeroy, 2025-03-14:
Any updates to this? Would love to see this