SasaKaranovic/OpenFanController #3: Portainer image build fail

CzarofAK, 2024-02-27:
I cant build an docker image within portainer. in pure docker it works. find below the log:
```
Step 1/16 : FROM ubuntu:22.04

 ---> 3db8720ecbf5

Step 2/16 : LABEL maintainer="sasa@karanovic.ca"

 ---> Using cache

 ---> c57be625cc03

Step 3/16 : LABEL version="0.1"

 ---> Using cache

 ---> 2f947d414698

Step 4/16 : LABEL description="OpenFan Controller Web GUI and Server docker image"

 ---> Using cache

 ---> b83f26ff6448

Step 5/16 : LABEL org.opencontainers.image.source=https://github.com/sasakaranovic/openfancontroller

 ---> Using cache

 ---> ccd81598a299

Step 6/16 : ARG DEBIAN_FRONTEND=noninteractive

 ---> Using cache

 ---> b0ba4b40a52d

Step 7/16 : RUN apt update

 ---> Using cache

 ---> 321e793a5359

Step 8/16 : RUN apt install -y python3 python3-pip

 ---> Using cache

 ---> 0fc26f98c6a2

Step 9/16 : RUN rm -rf /var/lib/apt/lists/*

 ---> Using cache

 ---> a49f3fecda59

Step 10/16 : RUN apt clean

 ---> Using cache

 ---> dbcf62869df5

Step 11/16 : ADD Software/Python /mnt/OpenFan

ADD failed: file not found in build context or excluded by .dockerignore: stat Software/Python: file does not exist
```

SasaKaranovic, 2024-02-28:
There is a docker image available through ghcr.io
Can you try using that one instead of building a new one?

https://github.com/SasaKaranovic/OpenFanController/pkgs/container/openfancontroller

SasaKaranovic, 2024-02-29:
@CzarofAK please let me know if this works for you and we can close this issue or if we need to figure out a different solution.

closed 2024-03-06 by SasaKaranovic