SasaKaranovic/OpenFanController #8: API bug

EinRainerZufall, 2024-05-16:
I noticed that when setting the RPM the fan index is based on 0.

The first fan:
`http://192.168.0.11:3210/api/v0/fan/0/rpm?value=1000`

The tenth fan:
`http://192.168.0.11:3210/api/v0/fan/9/rpm?value=1000`

And if you select 10, the following error occurs:
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.10/dist-packages/tornado/web.py", line 1790, in _execute
>     result = await result
>   File "/usr/local/lib/python3.10/dist-packages/tornado/web.py", line 2688, in get
>     self.absolute_path = self.validate_absolute_path(self.root, absolute_path)
>   File "/usr/local/lib/python3.10/dist-packages/tornado/web.py", line 2900, in validate_absolute_path
>     raise HTTPError(404)
> tornado.web.HTTPError: HTTP 404: Not Found

&nbsp;
&nbsp;

And when setting the PWM, the same error occurs:
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.10/dist-packages/tornado/web.py", line 1790, in _execute
>     result = await result
>   File "/usr/local/lib/python3.10/dist-packages/tornado/web.py", line 2688, in get
>     self.absolute_path = self.validate_absolute_path(self.root, absolute_path)
>   File "/usr/local/lib/python3.10/dist-packages/tornado/web.py", line 2900, in validate_absolute_path
>     raise HTTPError(404)
> tornado.web.HTTPError: HTTP 404: Not Found

EinRainerZufall, 2024-05-16:
The first error may be intentional and only the docu is wrong? Because the docu says that the index is `1-10`.

SasaKaranovic, 2024-05-17:
Hey @EinRainerZufall !

Thank you for catching this! You are absolutely right. There is a typo in the documentation.
The fan indexing should range from 0-9 instead of 1-10... I will fix that now.

https://github.com/SasaKaranovic/OpenFanController/blob/master/Software/Python/webserver.py#L242,L244

closed 2024-05-17 by SasaKaranovic

EinRainerZufall, 2024-05-17:
Hi @SasaKaranovic 

I continued testing today and found the error for setting a PWM value in the code.

According to the docu:
`http://192.168.0.11:3210/api/v0/fan/6/pwm?value=100`

but it is actually programmed:
`http://192.168.0.11:3210/api/v0/fan/6/set?value=100`

&nbsp;
&nbsp;
I think `pwm` would be better for clarity, but that would be a breaking change

SasaKaranovic, 2024-05-18:
Hey @EinRainerZufall !

Thank you for bringing this up. Another typo in the documentation that should be fixed now.

And I agree, `pwm` would make more sense for clarity but I am also reluctant to introduce breaking changes now...
Maybe I can add support for both `pwm` and `set` and then "phase out" `set` over time.