SasaKaranovic/OpenFanController #9: Stop Fans possible?

Fabian-Schmidt, 2024-05-19:
Do I see it correctly that you always supply 12V to the fan?

Some Fan understand 0% PWM signal as a stop and others spin at minimal speed.
Without means of interrupting power to the fan it is not possible to guarantee to stop a fan for a silent mode.

SasaKaranovic, 2024-05-19:
Hey Fabian.

OpenFAN controls the fans via PWM signal.

So at 0% PWM the fan will be at whatever state the fan manufacturer designed as minimum speed. Which should be OFF but there are fan manufacturers that deviate from this.

closed 2024-06-01 by Fabian-Schmidt

LoLo2207, 2025-01-14:
Sorry for reviving this, could it be possible to add in a V4 some kind of relays or transistors to turn off each channel? Because there are fans that don't have the 0RPM mode

SasaKaranovic, 2025-01-18:
Hey @LoLo2207 

Thank you for the suggestion. I can't promise anything other than it's already on the feature "wish-list" for next iteration.

Just briefly explaining some reasoning why this is not already a feature;
Adding such feature is very simple but it does increase overall complexity and cost. And the latter would be passed on to everyone regardless if they really need that feature or "don't care" about it.

SasaKaranovic, 2025-01-18:
For anyone reading this issue in the future and wondering why 0% PWM makes some fans stop and some are still spinning, here is the original [Intel 4-pin PWM specification](https://www.konilabs.net/docs/standards/fan/intel_4wire_pwm_fans_specs_rev1_2.pdf).

On `page 12`, `Figure 2 Fan Speed Response to PWM Control Input Signal` it shows that PWM below 20% is lebeled as "undetermined".
This usually leaves fan manufacturer to choose between having their fan stop completely (0 RPM) below certain PWM or have them run at some minimum (non-zero) RPM.