SasaKaranovic/OpenFanController #12: Fan alias feature

Th3M1k3y, 2024-05-31:
It would be nice to have the ability to rename fans in the UI.

Maybe something like Alias <sub>(FAN #1)</sub>

Control through the API remain as it is currently.

SasaKaranovic, 2024-06-01:
Thanks for the suggestion! I like this idea.

lord-xeon, 2024-06-02:
I also would like this feature.  I started to build something to keep track of the same for me.  While it's probably not a problem for everyone, it's nice to know `Fan 6` is the one on the top of the case.

SasaKaranovic, 2024-06-02:
@Th3M1k3y @lord-xeon 

I have just created a new branch https://github.com/SasaKaranovic/OpenFanController/tree/sk_general_improvements that contains API paths that allow setting of fan alias (name).

The UI is also updated to show aliases (if set).
To set an alias, you would double-click on the fan name and the pop-up will appear asking for new fan name.

If you have the chance, please give it a try and let me know if you have any suggestions.
I will take couple of days to test and make sure there are no issues or regressions and then if everything looks alright merge it into master.

Th3M1k3y, 2024-06-02:
I am running the controller in the unraid docker, not sure how to go about testing this.

SasaKaranovic, 2024-06-02:
>  I am running the controller in the unraid docker, not sure how to go about testing this.

You would have to run it manually or build a new docker image.
It's ok if it's not convenient for you, I will release a new docker image once it's tested and then you can just update your container.

SasaKaranovic, 2024-06-21:
[image]

This feature should be live now in the release branch.
If you update (force-update) your docker image, you should have this option.
Double-click on the fan name will show a prompt to enter a new fan name.

Let me know how it works and if you encounter any issues.

closed 2024-07-28 by github-actions[bot]