SasaKaranovic/OpenFanController #15: Enclosure can't close

Th3M1k3y, 2024-06-02:
A capacitor is too tall, so the lid can't be closed.

I added a hole so the capacitor could go through. With this the surface of the capacitor sits 0.1mm under the surface of the lid.

[image]

SasaKaranovic, 2024-06-02:
Thank you @Th3M1k3y !

I have been informed that the PCB shop used a slightly taller capacitor in this batch.
As the result it is colliding with the enclosure.
I will update the enclosure design to make it taller so it fits the board nicely.

SasaKaranovic, 2024-06-02:
@Th3M1k3y 
I have just updated the enclosure by increasing the height by 2mm.
If you want you can print the new enclosure and give it a try.
If it works well for you too, let me know and we can close this issue.

closed 2024-06-17 by SasaKaranovic