SasaKaranovic/OpenFanController #18: Not working with Fan Control software

RysatkoCZ, 2024-06-21:
Hello. 
First, thank you for this awesome project.
I wanted to increase fan ports in my mITX system so I built OpenFan Controller. I am using Fan Control to control fans in my system. But after connecting the controller to the motherboard the Fan Control stops updating the sensors until I disconnect the controller. When I try to restart the Fan Control it will not open until the controller is disconnected. I am using Fan Control V194, FanControl.OpenFAN plugin v20240129 and OpenFan firmware 2023-09-29_OpenFAN_FW_01.uf2.

SasaKaranovic, 2024-06-22:
Hey @RysatkoCZ !

I assume this is a custom built boards?
Did you assemble it yourself or did you have a PCB shop do it for you?

OpenFAN works with FanControl. I have it running on my PC (just updated to v194 from v193) and have multiple other people report that they are using it with FanControl using the provided plugin.

This means that most likely there is an issue with your board.

Have you double checked your hardware?
For example the white LED should be blinking when unit is programmed.
You can manually send commands over serial to check if your board is receiving commands properly and checking response.
You can run the provided python software and check the logs if there are any errors/issues reported.

RysatkoCZ, 2024-06-23:
The board was assembled by PCBway. The LED is blinking slowly after firmware flashing.
I am trying to run the python software but still get error when building it.

```
PS C:\Users\Pavel\Downloads\OpenFanController-master\Software\Python> pip install -r requirements.txt
Collecting tornado>=6.3.3 (from -r requirements.txt (line 1))
  Using cached tornado-6.4.1-cp38-abi3-win_amd64.whl.metadata (2.6 kB)
Collecting argparse (from -r requirements.txt (line 2))
  Using cached argparse-1.4.0-py2.py3-none-any.whl.metadata (2.8 kB)
Collecting pyserial==3.5 (from -r requirements.txt (line 3))
  Using cached pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Collecting ruamel.yaml==0.17.21 (from -r requirements.txt (line 4))
  Using cached ruamel.yaml-0.17.21-py3-none-any.whl.metadata (13 kB)
Collecting ruamel.yaml.clib==0.2.7 (from -r requirements.txt (line 5))
  Using cached ruamel.yaml.clib-0.2.7.tar.gz (182 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Using cached ruamel.yaml-0.17.21-py3-none-any.whl (109 kB)
Using cached tornado-6.4.1-cp38-abi3-win_amd64.whl (438 kB)
Using cached argparse-1.4.0-py2.py3-none-any.whl (23 kB)
Building wheels for collected packages: ruamel.yaml.clib
  Building wheel for ruamel.yaml.clib (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for ruamel.yaml.clib (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [119 lines of output]
      <string>:81: DeprecationWarning: ast.Str is deprecated and will be removed in Python 3.14; use ast.Constant instead
      <string>:81: DeprecationWarning: ast.Num is deprecated and will be removed in Python 3.14; use ast.Constant instead
      <string>:81: DeprecationWarning: ast.Bytes is deprecated and will be removed in Python 3.14; use ast.Constant instead
      <string>:81: DeprecationWarning: ast.NameConstant is deprecated and will be removed in Python 3.14; use ast.Constant instead
      <string>:103: DeprecationWarning: ast.Str is deprecated and will be removed in Python 3.14; use ast.Constant instead
      <string>:107: DeprecationWarning: ast.Bytes is deprecated and will be removed in Python 3.14; use ast.Constant instead
      <string>:109: DeprecationWarning: ast.Num is deprecated and will be removed in Python 3.14; use ast.Constant instead
      <string>:119: DeprecationWarning: ast.NameConstant is deprecated and will be removed in Python 3.14; use ast.Constant instead
      <string>:106: DeprecationWarning: Attribute s is deprecated and will be removed in Python 3.14; use value instead
      <string>:110: DeprecationWarning: Attribute n is deprecated and will be removed in Python 3.14; use value instead
      test_ruamel_yaml.c
      Vytv ýˇ se k˘d.
      Vytv ýenˇ k˘du se dokonźilo.
      C:\Users\Pavel\AppData\Local\Temp\pip-build-env-uosez9pj\overlay\Lib\site-packages\setuptools\dist.py:681: SetuptoolsDeprecationWarning: The namespace_packages parameter is deprecated.
      !!

              ********************************************************************************
              Please replace its usage with implicit namespaces (PEP 420).

              See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages for details.
              ********************************************************************************

      !!
        ep.load()(self, ep.name, value)
      sys.argv ['setup.py', 'bdist_wheel', '--dist-dir', 'C:\\Users\\Pavel\\AppData\\Local\\Temp\\pip-wheel-um2kqstr\\.tmp-zde2ynnq']
      test compiling C:\Users\Pavel\AppData\Local\Temp\tmp_ruamel_x29afh1w\test_ruamel_yaml.c -> test_ruamel_yaml OK
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build\lib.win-amd64-cpython-312
      creating build\lib.win-amd64-cpython-312\ruamel
      copying .ruamel\__init__.py -> build\lib.win-amd64-cpython-312\ruamel
      creating build\lib.win-amd64-cpython-312\ruamel\yaml
      copying .ruamel\yaml\__init__.py -> build\lib.win-amd64-cpython-312\ruamel\yaml
      creating build\lib.win-amd64-cpython-312\ruamel\yaml\clib
      copying .\__init__.py -> build\lib.win-amd64-cpython-312\ruamel\yaml\clib
      copying .\LICENSE -> build\lib.win-amd64-cpython-312\ruamel\yaml\clib
      running build_ext
      building '_ruamel_yaml' extension
      creating build\temp.win-amd64-cpython-312
[... 60 lines cut ...]
      _ruamel_yaml.c(27085): error C2039: curexc_traceback nenˇ źlenem: _ts.
      C:\Program Files\Python312\include\cpython/pystate.h(115): note: viz deklarace: _ts
      _ruamel_yaml.c(27624): warning C4013: _PyUnicode_Ready nedefinovan‚; pýedpokl d  se extern vracejˇcˇ int.
      _ruamel_yaml.c(28302): error C2039: ob_digit nenˇ źlenem: _longobject.
      C:\Program Files\Python312\include\cpython/longintrepr.h(87): note: viz deklarace: _longobject
      _ruamel_yaml.c(28357): error C2039: ob_digit nenˇ źlenem: _longobject.
      C:\Program Files\Python312\include\cpython/longintrepr.h(87): note: viz deklarace: _longobject
      _ruamel_yaml.c(28498): error C2039: ob_digit nenˇ źlenem: _longobject.
      C:\Program Files\Python312\include\cpython/longintrepr.h(87): note: viz deklarace: _longobject
      _ruamel_yaml.c(28553): error C2039: ob_digit nenˇ źlenem: _longobject.
[... cut ...]
      C:\Program Files\Python312\include\cpython/longintrepr.h(87): note: viz deklarace: _longobject
      error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.40.33807\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for ruamel.yaml.clib
Failed to build ruamel.yaml.clib
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (ruamel.yaml.clib)
```

SasaKaranovic, 2024-06-23:
It looks like installation of `ruamel.yaml` module failed on your system.
You try to fix it after which you should be able to run the software. Alternatively you can try running the docker image to test if your board was assembled properly.

RysatkoCZ, 2024-06-25:
I was not able to run it even after I installed ruamel.yaml separately. So I tried Docker Desktop. I have absolutely no expirience with it. Now I get this error:
```
PS C:\Users\Pavel> docker run --name OpenFAN -p 3000:3000 --device=/dev/ttyACM0 -e OPENFANCOMPORT=/dev/ttyACM0 sasakaranovic/openfan:latest
docker: Error response from daemon: error gathering device information while adding custom device "/dev/ttyACM0": no such file or directory.
```

When trying to start the container in the application, this appears in log:

```
2024-06-25 19:30:45 Jun 25 2024 17:30:45.974 pid:7 base_logger.py            59 INFO  Logging level: INFO
2024-06-25 19:30:45 Jun 25 2024 17:30:45.980 pid:7 config.py                 49 DEBUG Server config:
2024-06-25 19:30:45 Jun 25 2024 17:30:45.980 pid:7 config.py                 58 DEBUG {'hostname': 'localhost', 'port': 3000, 'communication_timeout': 1}
2024-06-25 19:30:45 Jun 25 2024 17:30:45.980 pid:7 config.py                 60 DEBUG Hardware config:
2024-06-25 19:30:45 Jun 25 2024 17:30:45.980 pid:7 config.py                 69 DEBUG {'port': None}
2024-06-25 19:30:45 Jun 25 2024 17:30:45.980 pid:7 config.py                 71 DEBUG Fan profile config:
2024-06-25 19:30:45 Jun 25 2024 17:30:45.980 pid:7 config.py                 80 DEBUG {'1000 RPM': {'type': 'rpm', 'values': [1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000], 'name': '1000 RPM'}, 'MadMax': {'type': 'PWM', 'values': [100, 100, 100, 100, 100, 100, 100, 100, 100, 100], 'name': 'MadMax'}, '1200RPM': {'type': 'RPM', 'values': [1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200], 'name': '1200RPM'}}
2024-06-25 19:30:45 Jun 25 2024 17:30:45.980 pid:7 config.py                 83 DEBUG Fan aliases:
2024-06-25 19:30:45 Jun 25 2024 17:30:45.980 pid:7 config.py                 92 DEBUG {0: 'Fan #1', 1: 'Fan #2', 2: 'Fan #3', 3: 'Fan #4', 4: 'Fan #5', 5: 'Fan #6', 6: 'Fan #7', 7: 'Fan #8', 8: 'Fan #9', 9: 'Fan #10'}
2024-06-25 19:30:45 Jun 25 2024 17:30:45.980 pid:7 FanCommander.py           98 DEBUG Searching for COM port with VID:0x2E8A and PID:0x000A
2024-06-25 19:30:45 Jun 25 2024 17:30:45.980 pid:7 webserver.py             299 ERROR Could not find Fan controller. Please make sure it's plugged in and (if necessary) drivers are installed.
2024-06-25 19:30:45 Jun 25 2024 17:30:45.980 pid:7 webserver.py             373 ERROR OpenFan API service crashed during setup.
2024-06-25 19:30:45 Traceback (most recent call last):
2024-06-25 19:30:45   File "/mnt/OpenFan/webserver.py", line 371, in main
2024-06-25 19:30:45     FAN_API_Service().run_forever()
2024-06-25 19:30:45   File "/mnt/OpenFan/webserver.py", line 300, in __init__
2024-06-25 19:30:45     raise Exception("Could not find Fan controller. Please make sure it's plugged in and (if necessary) drivers are installed.")
2024-06-25 19:30:45 Exception: Could not find Fan controller. Please make sure it's plugged in and (if necessary) drivers are installed.
```

So I belive the board is dead?

SasaKaranovic, 2024-06-26:
It looks like you are on Windows, so find which COM port is your board, and then replace `/dev/ttyACM0` in the Docker run command with that. So for example if it's `COM1` your docker run command would look like `docker run --name OpenFAN -p 3000:3000 --device=COM1 -e OPENFANCOMPORT=COM1 sasakaranovic/openfan:latest`

Let me know how it goes.
Also your board might be faulty/not-assembled properly by the PCB shop that you used. But maybe you could still salvage it.

RysatkoCZ, 2024-06-26:
When I connect the board USB Serial Device COM8 and USB Serial Device COM9 appears in the Device Manager. I tried both of them, but I get this:

```
PS C:\Users\Pavel> docker run --name OpenFAN -p 3000:3000 --device=COM8 -e OPENFANCOMPORT=COM8 sasakaranovic/openfan:latest
docker: COM8 is not an absolute path.
See 'docker run --help'.
PS C:\Users\Pavel> docker run --name OpenFAN -p 3000:3000 --device=COM9 -e OPENFANCOMPORT=COM9 sasakaranovic/openfan:latest
docker: COM9 is not an absolute path.
See 'docker run --help'.
```

SasaKaranovic, 2024-06-28:
You are on Windows so accessing the COM port from within docker might be a bit different.

You can try running the container in privileged mode. Something like this

`docker run --privileged --name OpenFAN -p 3000:3000 --device=COM8 -e OPENFANCOMPORT=COM8 sasakaranovic/openfan:latest `

p.s. You might need to remove the old container first with `docker rm OpenFAN`

RysatkoCZ, 2024-06-28:
Still the same issue. I have googled a bit and tried to use `--device=/dev/COM8` and I get following:

```
docker run --privileged --name OpenFAN -p 3000:3000 --device=/dev/COM8 -e OPENFANCOMPORT=/dev/COM8 sasakaranovic/openfan:latest
Jun 28 2024 08:12:32.204 pid:7 base_logger.py            59 INFO  Logging level: INFO
Jun 28 2024 08:12:32.210 pid:7 config.py                 49 DEBUG Server config:
Jun 28 2024 08:12:32.210 pid:7 config.py                 58 DEBUG {'hostname': 'localhost', 'port': 3000, 'communication_timeout': 1}
Jun 28 2024 08:12:32.210 pid:7 config.py                 60 DEBUG Hardware config:
Jun 28 2024 08:12:32.210 pid:7 config.py                 69 DEBUG {'port': None}
Jun 28 2024 08:12:32.210 pid:7 config.py                 71 DEBUG Fan profile config:
Jun 28 2024 08:12:32.210 pid:7 config.py                 80 DEBUG {'1000 RPM': {'type': 'rpm', 'values': [1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000], 'name': '1000 RPM'}, 'MadMax': {'type': 'PWM', 'values': [100, 100, 100, 100, 100, 100, 100, 100, 100, 100], 'name': 'MadMax'}, '1200RPM': {'type': 'RPM', 'values': [1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200], 'name': '1200RPM'}}
Jun 28 2024 08:12:32.210 pid:7 config.py                 83 DEBUG Fan aliases:
Jun 28 2024 08:12:32.210 pid:7 config.py                 92 DEBUG {0: 'Fan #1', 1: 'Fan #2', 2: 'Fan #3', 3: 'Fan #4', 4: 'Fan #5', 5: 'Fan #6', 6: 'Fan #7', 7: 'Fan #8', 8: 'Fan #9', 9: 'Fan #10'}
Jun 28 2024 08:12:32.210 pid:7 webserver.py             295 INFO  Using COM port `/dev/COM8` (specified in `OPENFANCOMPORT` env variable).
Jun 28 2024 08:12:32.210 pid:7 webserver.py             304 INFO  Fan Controller port: /dev/COM8
Jun 28 2024 08:12:32.212 pid:7 serial_driver.py          87 DEBUG Searching for COM port `/dev/COM8`
Jun 28 2024 08:12:32.212 pid:7 webserver.py             373 ERROR OpenFan API service crashed during setup.
Traceback (most recent call last):
  File "/mnt/OpenFan/webserver.py", line 371, in main
    FAN_API_Service().run_forever()
  File "/mnt/OpenFan/webserver.py", line 305, in __init__
    self.fan_commander = FanCommander(self.serialPort)
  File "/mnt/OpenFan/FanCommander.py", line 10, in __init__
    super(FanCommander, self).__init__(port_info, timeout=2)
  File "/mnt/OpenFan/serial_driver.py", line 28, in __init__
    raise TypeError("The port_info must be of type {} (given: {})".format(self.__class__, _lpc.ListPortInfo))
TypeError: The port_info must be of type <class 'FanCommander.FanCommander'> (given: <class 'serial.tools.list_ports_common.ListPortInfo'>)
```

azzajess, 2024-06-30:
Sorry to semi hijack this issue but Im getting the same identical problem to @RysatkoCZ with my PCBway built board. The only self assembly was the sata connector. I am able to flash the firmware onto the board when holding down boot and dragging the firmware in. It reboots and the ACT is flashing.

Trying the python software I ran into similar issues with @RysatkoCZ when running `pip install -r requirements.txt`
This was on the latest python, so I just downgraded to 3.9 and was able to install dependence without issue.
However when running the webserver.py it hangs a bit on the com port 4 of the openfan controller. then after a while it closes the script with an error.
```
C:\Users\PC-User\Downloads\OpenFanController-master\OpenFanController-master\Software\Python>python webserver.py
←[0;36mJun 30 2024 19:17:45.893←[0m ←[1;31mpid:15832←[0m ←[0;34mbase_logger.py            59←[0m ←[1;33mINFO ←[0m ←[0;32mLogging level: INFO←[0m
←[0;36mJun 30 2024 19:17:45.901←[0m ←[1;31mpid:15832←[0m ←[0;34mconfig.py                 49←[0m ←[1;33mDEBUG←[0m ←[0;32mServer config:←[0m
←[0;36mJun 30 2024 19:17:45.901←[0m ←[1;31mpid:15832←[0m ←[0;34mconfig.py                 58←[0m ←[1;33mDEBUG←[0m ←[0;32m{'hostname': 'localhost', 'port': 3000, 'communication_timeout': 1}←[0m
←[0;36mJun 30 2024 19:17:45.902←[0m ←[1;31mpid:15832←[0m ←[0;34mconfig.py                 60←[0m ←[1;33mDEBUG←[0m ←[0;32mHardware config:←[0m
←[0;36mJun 30 2024 19:17:45.902←[0m ←[1;31mpid:15832←[0m ←[0;34mconfig.py                 69←[0m ←[1;33mDEBUG←[0m ←[0;32m{'port': None}←[0m
←[0;36mJun 30 2024 19:17:45.902←[0m ←[1;31mpid:15832←[0m ←[0;34mconfig.py                 71←[0m ←[1;33mDEBUG←[0m ←[0;32mFan profile config:←[0m
←[0;36mJun 30 2024 19:17:45.903←[0m ←[1;31mpid:15832←[0m ←[0;34mconfig.py                 80←[0m ←[1;33mDEBUG←[0m ←[0;32m{'1000 RPM': {'type': 'rpm', 'values': [1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000], 'name': '1000 RPM'}, 'MadMax': {'type': 'PWM', 'values': [100, 100, 100, 100, 100, 100, 100, 100, 100, 100], 'name': 'MadMax'}, '1200RPM': {'type': 'RPM', 'values': [1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200], 'name': '1200RPM'}}←[0m
←[0;36mJun 30 2024 19:17:45.903←[0m ←[1;31mpid:15832←[0m ←[0;34mconfig.py                 83←[0m ←[1;33mDEBUG←[0m ←[0;32mFan aliases:←[0m
←[0;36mJun 30 2024 19:17:45.903←[0m ←[1;31mpid:15832←[0m ←[0;34mconfig.py                 92←[0m ←[1;33mDEBUG←[0m ←[0;32m{0: 'Fan #1', 1: 'Fan #2', 2: 'Fan #3', 3: 'Fan #4', 4: 'Fan #5', 5: 'Fan #6', 6: 'Fan #7', 7: 'Fan #8', 8: 'Fan #9', 9: 'Fan #10'}←[0m
←[0;36mJun 30 2024 19:17:45.909←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py           98←[0m ←[1;33mDEBUG←[0m ←[0;32mSearching for COM port with VID:0x2E8A and PID:0x000A←[0m
←[0;36mJun 30 2024 19:17:45.909←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          100←[0m ←[1;33mDEBUG←[0m ←[0;32mCOM1←[0m
←[0;36mJun 30 2024 19:17:45.910←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          101←[0m ←[1;33mDEBUG←[0m ←[0;32m       Product: None←[0m
←[0;36mJun 30 2024 19:17:45.910←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          102←[0m ←[1;33mDEBUG←[0m ←[0;32m       Desc: Communications Port (COM1)←[0m
←[0;36mJun 30 2024 19:17:45.910←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          103←[0m ←[1;33mDEBUG←[0m ←[0;32m       SN: None←[0m
←[0;36mJun 30 2024 19:17:45.910←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          104←[0m ←[1;33mDEBUG←[0m ←[0;32m       VID:None PID:None←[0m
←[0;36mJun 30 2024 19:17:45.911←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          105←[0m ←[1;33mDEBUG←[0m ←[0;32m       Location: None←[0m
←[0;36mJun 30 2024 19:17:45.911←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          106←[0m ←[1;33mDEBUG←[0m ←[0;32m       Interface: None←[0m
←[0;36mJun 30 2024 19:17:45.911←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          100←[0m ←[1;33mDEBUG←[0m ←[0;32mCOM4←[0m
←[0;36mJun 30 2024 19:17:45.912←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          101←[0m ←[1;33mDEBUG←[0m ←[0;32m       Product: None←[0m
←[0;36mJun 30 2024 19:17:45.912←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          102←[0m ←[1;33mDEBUG←[0m ←[0;32m       Desc: USB Serial Device (COM4)←[0m
←[0;36mJun 30 2024 19:17:45.912←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          103←[0m ←[1;33mDEBUG←[0m ←[0;32m       SN: DE627CA44F553A37←[0m
←[0;36mJun 30 2024 19:17:45.913←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          104←[0m ←[1;33mDEBUG←[0m ←[0;32m       VID:11914 PID:10←[0m
←[0;36mJun 30 2024 19:17:45.913←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          105←[0m ←[1;33mDEBUG←[0m ←[0;32m       Location: None←[0m
←[0;36mJun 30 2024 19:17:45.913←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          106←[0m ←[1;33mDEBUG←[0m ←[0;32m       Interface: None←[0m
←[0;36mJun 30 2024 19:17:45.913←[0m ←[1;31mpid:15832←[0m ←[0;34mFanCommander.py          108←[0m ←[1;33mDEBUG←[0m ←[0;32mUsing 'USB Serial Device (COM4)' as Fan Controller COM port←[0m
←[0;36mJun 30 2024 19:17:45.914←[0m ←[1;31mpid:15832←[0m ←[0;34mwebserver.py             302←[0m ←[1;33mINFO ←[0m ←[0;32mUsing COM port `COM4 - USB Serial Device (COM4)` (Found through USB VID:PID).←[0m
←[0;36mJun 30 2024 19:17:45.914←[0m ←[1;31mpid:15832←[0m ←[0;34mwebserver.py             304←[0m ←[1;33mINFO ←[0m ←[0;32mFan Controller port: COM4 - USB Serial Device (COM4)←[0m
[... cut ...]
    raise SerialException(
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: OSError(22, 'The semaphore timeout period has expired.', None, 121)
```

After a few tries and restarts, I can get a webserver up but it doesnt display any RPM values. Its like its still not communicating properly? I can drag the rpm or percentage slider but it doesnt apply to the fans ether.

Thats where im at, at the moment. Ill update this post if I discover anything new...

RysatkoCZ, 2024-06-30:
Ok. I have also downgraded to Python 3.9 and I am in. But same as @azzajess , no fan control is available in the web interface.

Maybe it helps with troubleshooting, I have uploaded [BOM](https://github.com/user-attachments/files/16045762/BOM.xls) from which my board was assembled. Maybe I have messed it up somehow.

```
PS C:\Software\Python> python webserver.py
Jun 30 2024 20:26:50.618 pid:39688 base_logger.py            59 INFO  Logging level: INFO
Jun 30 2024 20:26:50.625 pid:39688 config.py                 46 DEBUG Server config:
Jun 30 2024 20:26:50.625 pid:39688 config.py                 55 DEBUG {'hostname': 'localhost', 'port': 3000, 'communication_timeout': 1}
Jun 30 2024 20:26:50.625 pid:39688 config.py                 57 DEBUG Hardware config:
Jun 30 2024 20:26:50.625 pid:39688 config.py                 66 DEBUG {'port': None}
Jun 30 2024 20:26:50.625 pid:39688 config.py                 68 DEBUG Fan profile config:
Jun 30 2024 20:26:50.625 pid:39688 config.py                 77 DEBUG {'1000 RPM': {'type': 'rpm', 'values': [1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000], 'name': '1000 RPM'}, 'MadMax': {'type': 'PWM', 'values': [100, 100, 100, 100, 100, 100, 100, 100, 100, 100]}, '1200RPM': {'type': 'RPM', 'values': [1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200]}}
Jun 30 2024 20:26:50.633 pid:39688 FanCommander.py           99 DEBUG Searching for COM port with VID:0x2E8A and PID:0x000A
Jun 30 2024 20:26:50.633 pid:39688 FanCommander.py          101 DEBUG COM8
Jun 30 2024 20:26:50.633 pid:39688 FanCommander.py          102 DEBUG   Product: None
Jun 30 2024 20:26:50.633 pid:39688 FanCommander.py          103 DEBUG   Desc: Sériové zařízení USB (COM8)
Jun 30 2024 20:26:50.633 pid:39688 FanCommander.py          104 DEBUG   SN: DE6048D8577A3529
Jun 30 2024 20:26:50.633 pid:39688 FanCommander.py          105 DEBUG   VID:11914 PID:10
Jun 30 2024 20:26:50.634 pid:39688 FanCommander.py          106 DEBUG   Location: 1-12:x.0
Jun 30 2024 20:26:50.634 pid:39688 FanCommander.py          107 DEBUG   Interface: None
Jun 30 2024 20:26:50.634 pid:39688 FanCommander.py          109 DEBUG Using 'Sériové zařízení USB (COM8) ' as Fan Controller COM port
Jun 30 2024 20:26:50.634 pid:39688 webserver.py             231 INFO  Using COM port `COM8 - Sériové zařízení USB (COM8) ` (Found through USB VID:PID).
Jun 30 2024 20:26:50.634 pid:39688 webserver.py             233 INFO  Fan Controller port: COM8 - Sériové zařízení USB (COM8)
Jun 30 2024 20:26:50.640 pid:39688 serial_driver.py          43 DEBUG Serial driver initialized with
Jun 30 2024 20:26:50.640 pid:39688 serial_driver.py          44 DEBUG -- flush_on_write: 'True'
Jun 30 2024 20:26:50.640 pid:39688 serial_driver.py          45 DEBUG -- timeout: '2'
Jun 30 2024 20:26:50.640 pid:39688 webserver.py             258 INFO  Karanovic Research OpenFan - Starting API server
Jun 30 2024 20:26:50.640 pid:39688 webserver.py             263 INFO  Listening on port 3000
Jun 30 2024 20:27:16.424 pid:39688 webserver.py             189 DEBUG Requesting: None
Jun 30 2024 20:27:16.785 pid:39688 webserver.py              45 INFO  Request:FANPROFILE_LIST
Jun 30 2024 20:27:17.786 pid:39688 webserver.py             129 INFO  Request:STATUS
Jun 30 2024 20:27:17.787 pid:39688 FanCommander.py           24 DEBUG CMD:0 Data: None
Jun 30 2024 20:27:17.787 pid:39688 FanCommander.py           25 DEBUG Sending `>00`
```

[image]

SasaKaranovic, 2024-06-30:
> Still the same issue. I have googled a bit and tried to use --device=/dev/COM8

This will not work on Windows, please try `docker run --privileged --name OpenFAN -p 3000:3000 -e OPENFANCOMPORT=COM8 sasakaranovic/openfan:latest `

Your BOM is not the same as the one provided in the repo. I didn't do a deep dive but just from glancing at it resistors have worse tolerance, uses a different inductor on the switching regulator and so on. I can't say if this is causing issues or not, but also note that this is essentially a different BOM from the one that was provided.

From the logs you posted it seems that your PC is having issues communicating with the board which points to most likely a hardware issue with the board or/and cabling.
Can you try using the shortest USB cable you have (and good quality) to verify if you are maybe having signal integrity issues.
The design is fairly simple but at the same time it has some places for PCB shop to mess it up (ie using cheaper components, changing stackup, messing up impedances etc).

RysatkoCZ, 2024-06-30:
> This will not work on Windows, please try docker run --privileged --name OpenFAN -p 3000:3000 -e OPENFANCOMPORT=COM8 sasakaranovic/openfan:latest 

```
docker run --privileged --name OpenFAN -p 3000:3000 -e OPENFANCOMPORT=COM8 sasakaranovic/openfan:latest
Jun 30 2024 21:13:05.989 pid:7 base_logger.py            59 INFO  Logging level: INFO
Jun 30 2024 21:13:05.995 pid:7 config.py                 49 DEBUG Server config:
Jun 30 2024 21:13:05.995 pid:7 config.py                 58 DEBUG {'hostname': 'localhost', 'port': 3000, 'communication_timeout': 1}
Jun 30 2024 21:13:05.995 pid:7 config.py                 60 DEBUG Hardware config:
Jun 30 2024 21:13:05.995 pid:7 config.py                 69 DEBUG {'port': None}
Jun 30 2024 21:13:05.995 pid:7 config.py                 71 DEBUG Fan profile config:
Jun 30 2024 21:13:05.995 pid:7 config.py                 80 DEBUG {'1000 RPM': {'type': 'rpm', 'values': [1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000], 'name': '1000 RPM'}, 'MadMax': {'type': 'PWM', 'values': [100, 100, 100, 100, 100, 100, 100, 100, 100, 100], 'name': 'MadMax'}, '1200RPM': {'type': 'RPM', 'values': [1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200], 'name': '1200RPM'}}
Jun 30 2024 21:13:05.995 pid:7 config.py                 83 DEBUG Fan aliases:
Jun 30 2024 21:13:05.995 pid:7 config.py                 92 DEBUG {0: 'Fan #1', 1: 'Fan #2', 2: 'Fan #3', 3: 'Fan #4', 4: 'Fan #5', 5: 'Fan #6', 6: 'Fan #7', 7: 'Fan #8', 8: 'Fan #9', 9: 'Fan #10'}
Jun 30 2024 21:13:05.995 pid:7 webserver.py             295 INFO  Using COM port `COM8` (specified in `OPENFANCOMPORT` env variable).
Jun 30 2024 21:13:05.995 pid:7 webserver.py             304 INFO  Fan Controller port: COM8
Jun 30 2024 21:13:05.996 pid:7 serial_driver.py          87 DEBUG Searching for COM port `COM8`
Jun 30 2024 21:13:05.996 pid:7 webserver.py             373 ERROR OpenFan API service crashed during setup.
Traceback (most recent call last):
  File "/mnt/OpenFan/webserver.py", line 371, in main
    FAN_API_Service().run_forever()
  File "/mnt/OpenFan/webserver.py", line 305, in __init__
    self.fan_commander = FanCommander(self.serialPort)
  File "/mnt/OpenFan/FanCommander.py", line 10, in __init__
    super(FanCommander, self).__init__(port_info, timeout=2)
  File "/mnt/OpenFan/serial_driver.py", line 28, in __init__
    raise TypeError("The port_info must be of type {} (given: {})".format(self.__class__, _lpc.ListPortInfo))
TypeError: The port_info must be of type <class 'FanCommander.FanCommander'> (given: <class 'serial.tools.list_ports_common.ListPortInfo'>)
```

The only [BOM](https://github.com/user-attachments/files/16046244/2023-06-04_23-21_OpenFAN_PC_PI_BOM.xlsx) I could find in the repo does not have specified resistors or capacitors. That is why I pulled the values from Altrium and searched for alternatives.

[image]

RysatkoCZ, 2024-06-30:
I have tried several cables, to USB-A and to MB USB, the same outcome with all of them. I think I will cut my losses and order a new board from the shop.

azzajess, 2024-06-30:
I had to do the same thing and go through the atrium files and inspect and note each component from there and its values. Then made my own BOM and posted it as an issue. 
@SasaKaranovic do you have access to a BOM with all the values so that I can compare it with the BOM i made up and double check any differences?

SasaKaranovic, 2024-07-01:
For some reason last release package was never pushed to Github... Sorry about that!
I did push it manually right now and you should be able to compare the BOMs from the [release folder](https://github.com/SasaKaranovic/OpenFanController/tree/master/Hardware/Altium_Source_Files/Release/2023-10-28_2203_OpenFan_R03/BOM)

azzajess, 2024-07-01:
Hmmmmm ok. Comparing them both, i noticed one (so far) major difference. I noticed the same error on @RysatkoCZ Bom list.
The Altrium files stated that C2,C4,C6,C10,C11,C12,C13,C14,C15,C16,C17,C18,C19,C21,C22,C24 had a capacitance of 100nf (.1uf/100000pF). But your BOM states that its 10nF (0.01uF/10000pF). Im guessing this accounts for the major difference. The other differences I found were tolerances differences. Ill start by changing these to the correct value and see if that makes a difference.
It just had to be the the one with the most capacitors. 16*5 :(

SasaKaranovic, 2024-07-01:
Those capacitors are 100nF in the schematic so you should be good. I've updated the BOM to show 104 (100nF) instead of 103 (10nF).

azzajess, 2024-07-02:
Ok, good to know, thanks. I noticed that the bom also states that c7, c8 is listed as 7p in the bom, but says 8 in the schematic. But as far as I can tell there is no difference between the bom list, component wise, I made and the one provided. The only difference I can tell is the tolerance, but surely it should at least be functionality?

SasaKaranovic, 2024-07-03:
Those differences should not make a huge impact. From what you told me I would expect both of your boards to be functional or at least "partially" work.
Just a wild random suggestion but maybe double check your board for shorts/open or if resistor/capacitor got swapped... You could also check if there is a unreasonably high current draw.
Sorry but since I don't have your boards in front of me it's hard to give a more helpful suggestion.

azzajess, 2024-07-06:
Im going through the board now trying to fault find. I noticed in the schematic we have LED then Resistor then Ground from the pi gpio. So LED is next to ACT print. However on the images on youtube and repo/website its resistor next to act then led to ground. Can you confirm what its suppose to be? On the rp pico, they have resistor, then led to ground. But this shouldnt affect the board though right?

azzajess, 2024-07-21:
I have gone over the board with a fine tooth comb, measuring every component and still have 0 idea why its not working. 

Where do you get your boards from? Are the BOM/Gerber files exactly the same as the ones you provided to that company?
Do you know anyone who has successfully DIYed their own board using the repository?
I cant figure out what would be different. the main components are exacts versions listed in bom/schematic.

SasaKaranovic, 2024-07-22:
I did receive couple of emails to let me know that people have made their own DIY version of the board.
So I would say that (at least two) people have successfully made their own boards using this repository.

azzajess, 2024-07-24:
Fair enough. Im stumped on why there is an issue. 

The only thing that I can logically conclude to is that there is some sort of communication error between the PWM controllers the pico chip. Because both PC fan control software and the python web server software will not show the tach value. And it will not set the PWM values. 

Is there a way to debug via software whether or where the communication error occurs?

closed 2024-08-13 by github-actions[bot]