SasaKaranovic/OpenFanController #22: USB-C to use it as standalone?

MrSpoocy, 2024-08-18:
Hello Sasa,

I am currently looking for a fan controller for my server cabinet and later for my living room cabinet. Unfortunately it is not possible to use a SATA power connector. I have seen that you already had a version with a DC-Jack connector. My question is, is it possible to swap the Micro-USB for USB-C with PD? So that it can be used for programming and power supply?

I plan to use ESP-Home as the firmware, as this now also supports the RP2040.

Kind regards Manuel

SasaKaranovic, 2024-09-04:
Hey @MrSpoocy !

I have two new version of OpenFAN coming up soon. Both of them will be stand-alone and work over WiFi.

In the meantime, you can also buy DC-to-SATA power adapters (Quick search on AliExpress gives a ton of options, see [here](https://www.aliexpress.com/w/wholesale-dc-to-sata-power.html))

p.s. Regarding USB-C PD; I think that the cable/adapter you linked is still USB2.0 without PD, it just has USB-C connector, so that would not work with the USB-C PD. Also from what I could tell, for some reason 12V on the USB-C PD is not really that popular. The 9V and 15V are way more common, but obviously don't work that well for our use case.

p.s. Sorry about late reply. I was away on vacation. :)

osnabrugge, 2024-09-04:
@SasaKaranovic what about PoE?  Has there been enough interest to build support for it?  That might also be a good solution for  @MrSpoocy if you have a PoE switch in your cabinet.

@SasaKaranovic I almost was about to buy a unit or two today but came across your blog about these new models.  Also, I noticed we live in the same city, I assume your storefront is direct from China, there's no way that I can transact locally to avoid to currency exchange and long shipping times / fees?

SasaKaranovic, 2024-09-05:
Hey @osnabrugge!

The PoE idea was considered at some point but I'm not sure how ubiquitous PoE is these days and how willing people are to "waste"/allocate some of that PoE budget to the fans. Honestly it would not be too difficult to built the PoE version but I suspect it would bloat the price and also make the product (even) more niche than it is right now. :\

You are correct, boards are produced and shipped from China so unfortunately I don't have any stock with me. Only a few prototypes and "beta" units.

Manufacturing these boards (sourcing, fabrication, programming, testing, QA etc) takes "a long time" so I don't generally like to announce/hype things before everything is completed and I have them in stock and they are ready to go. But I guess I already spilled the beans on the two additional versions of the OpenFAN so I might as well give some context. :)

- Active: OpenFAN is the solution if you need to control up to 10x PWM fans in your PC/server using SATA power and control over USB (and/or network using the software). This solution will continue to exist along with the new ones since they have different use-cases.
- Being manufactured. Available: ~end of September 2024: OpenFAN Micro is a version that controls single 5V or 12V PWM fan over WiFi and is powered from USB (no PD required)
- Not being manufactured. Available: TBD: OpenFAN WiFi is a version that controls up to 5x 5V or 12V PWM fans over WiFi and is using DC barrel jack for power.

OpenFAN Micro is close to being released so I will soon make a video explaining how it works, where it can be used and so on. Similar to what I did with the original OpenFAN.

Feel free to email me if you need more information on these or have more questions. :)

MrSpoocy, 2024-09-05:
Oh, I think my link was a bit misleading, this cable should only be for people who want to connect the controller further to their mainboard.

However, I think that USB-C PD with a “step down” is better than a DC jack. Good USB-C power supply are much more plentiful and easier to find (https://www.amazon.de/dp/B0CFF813QC). If you opt for USB-C PD, it should be compatible with the full standard (e.g step up for a original https://www.amazon.de/dp/B0CM46P7MC).

osnabrugge, 2024-09-06:
# Power over Ethernet (PoE) and USB-C Power Delivery (PD) for Home Labs

@SasaKaranovic thanks for your hard work and awesome update! I know a large community that makes use of PoE in their home labs (i.e. SBCs, APs, Cameras) and I'd be happy to introduce you to the originator of this project/discord group, I showed him your video last week and he found your project quite interesting. The group contributes to a community driven template that automates deployment(s) of a GitOps workflow including the setup of a k3s cluster that could be a single node on an SBC or as many nodes and platforms as you desire (so long as master nodes deployed meets HA requirements for quorum) thus strongly supports true open-source and certainly prefer free/low-cost solutions.

It might be possible through add-on modules to satisfy alternate power sources while keeping the original fan controller kit unmodified. However, it depends on whether the base board is extensible and able to handle the additional wattage. Otherwise, a V2 base board would be required to still meet the project's goals of keeping base cost low.

Using Power over Ethernet (PoE) and USB-C Power Delivery (USB-C PD) in home lab environments has both advantages and disadvantages. Add-on modules could be designed for specific power sources and/or a universal add-on to support SATA, USB-C PD, and PoE to satisfy @MrSpoocy's idea as well.

## Pros, Cons, and Use Cases for Supporting PoE and USB-C PD

- **PoE**
  - **Advantages**:
    - Greater distance (up to 100 meters)
    - Centralized power delivery control through the switch
    - Single Ethernet cable for both power and data, simplifying remote management.
  - **Disadvantages**:
    - Requires detailed installation and configuration documentation
    - Requires safety measures due to varying power standards (15W - 100W) that are non-interoperable.
  - **Use Case**:
    - Ideal for environments with existing PoE switch deployments
    - Existing/Desired solution to centrally manage other PoE devices already (e.g., AP, SBC, Camera).

- **USB-C PD**
  - **Advantages**:
    - Higher wattage (100W - 260W)
    - Automatic detection of the USB-C version and its corresponding power delivery limits
    - Reduced need for additional documentation and safety measures
  - **Disadvantages**:
    - Limited to 3 meters
    - Requires multiple cables or Wi-Fi for remote management
    - Lacks centralized management for multiple devices
  - **Use Case**:
    - Suitable for smaller environments without the need for remote access or centralized management
    - Especially useful for modern desktops with internal USB-C headers.

## Next Steps

Instead of hijacking this thread any further (sorry @MrSpoocy but I hoped that this message helps amplify supporting your use case as well), should I open a new issue, contact you privately through a real-time channel of your preference or via email as you suggested?

SasaKaranovic, 2024-09-07:
@MrSpoocy I agree that USB-C PD (and potential step-down) is a good idea. 
However in practice I think it would be a mess due to different PD chargers have different set of voltages that they support.
This mess is created mainly because there is no "certification" for PD, and anyone who has any voltage other than 5V labels their charger as PD (even the cheap 5V/9V only phone chargers).

You could definitely get one that works well and has the voltage you need, but there is inevitably going to be a group of people who are going to be asking why their PD charger does not work. (So advertising it as USB-C PD could potentially be "misleading"?)

Also if USB-C is used for communication as well, then there is a problem if your motherboard/PC does not have PD, because then you can't power and communicate with the board using the same cable.

@osnabrugge Feel free to email me or open a new issue, either/both are fine with me.
For USB-C PD the above concern is the main issue. USB-C PD is not that well standardised (ie. which voltages must be supported) and could cause confusion/problems for some users.

The PoE version is interesting because there is one single cable for both power and data. 
But depending on the fans attached you could easily take 10-15W out of your PoE budget. 
Also there are different PoE standards that might increase the cost and complexity of the board if we opt to support all of them.

Keep in mind that WiFi edition already has only one cable for power and that's it.
And users could use a a cheap DC adaptor for power everything or even get a PoE-to-DC adaptor.

closed 2024-09-29 by github-actions[bot]