SasaKaranovic/OpenFanController #26: Fan Input extension

anthony0030, 2024-11-15:
A feature that I thought of is to use one or both of the expansions ports to be able to put a fan input from the motherboard. This will send data back to the motherboard emulating a fan and get data from the fan header on the motherboard to be able to set a custom fan speed from the data coming from the motherboard, especially useful if you only have one or two fan headers and want to control different fans.

SasaKaranovic, 2024-11-17:
Thanks for the suggestion @anthony0030 !

This is a neat idea, but I think it does not align with this project and it's somewhat redundant.
We want to decouple fan control from hardware, software/OS etc.

If I did not understand you suggestion, feel free to correct me. But this feature would create a dependency between OpenFAN and the motherboard and how the motherboard drives the fan, what software you have to use to create those fan curves etc.

With OpenFAN, the idea is that you have critical fan control (ie. CPU fan) connected directly to the motherboard and then everything else i controlled by the OpenFAN.

anthony0030, 2024-11-18:
Good morning @SasaKaranovic, 

Thank you for getting back to me.

I was unaware of:
> With OpenFAN, the idea is that you have critical fan control (ie. CPU fan) connected directly to the motherboard and then everything else i controlled by the OpenFAN.

I was trying to explain the functionality of the [na-fc1](https://noctua.at/en/na-fc1) by noctua.
It basically sits between the motherboard and the fans and you set the potentiometer to 50% and it will reduce the fan speed by that much while still providing fan data to the motherboard. There would be no additional software in my example just another option in your existing software to be able to accept the fan output from the motherboard and reduce it when it gets to the fan, Thanks again for your response and great videos!

closed 2024-11-20 by SasaKaranovic

anthony0030, 2024-11-20:
@SasaKaranovic thank you for taking the time to clarify this for me. 

I look forward to doing a project that uses this fan controller. It looks like an amazing project, and the PCB looks really cool.