SasaKaranovic/OpenFanController #29: custom default fan profile

matdibu, 2025-01-03:
Thank you for making this, it's working perfectly!

That being said, would it be possible to embed a custom default fan profile directly into the board?

I've figured out exactly where I like my fans to be at, but it's a bit annoying having to rely on the host OS (I dual boot, so I have to configure it twice) to apply that profile.

I'd be much happier if I could flash/hardcode my settings as the default profile.

SasaKaranovic, 2025-01-03:
Hey @matdibu !

Ideally you would not want to run the OpenFAN Controller in a "headless" (without PC/SW) mode.

If you are dual booting, you could share the config file in both OS-es, for example share the SW directory or just the config file.
And then have the software re-apply the profile on boot without any interaction on your side.

If you really don't want to run the software, you could create a script that on-boot sends fan profile and that's it.
It's up to you how you want to use your board, but I would highly recommend running the software and actively adjusting the fan speed based on temperature and/or load. :)

matdibu, 2025-01-09:
This is what I settled on: https://codeberg.org/mateidibu/nix-config/src/commit/1cb80413e8337da7e914414f50a915c73c36c45e/hosts/x570/services/containers/openfan.nix

I start the container and then use the API to create and apply a profile

closed 2025-01-09 by matdibu