SasaKaranovic/OpenFanController #30: Some questions. Can it be used wirelessly and max amps

LoLo2207, 2025-01-14:
Hi, I just found about this controller thanks to a Reddit [reply ](https://www.reddit.com/r/esp32/comments/1i0i4q6/comment/m75omac/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button) because I was trying to do something similar myself but I have no experience designing PCBs

I have an external radiator and this would be perfect if I could use it wirelessly. Does it need to be always plugged in to the USB or the configuration website and FanControl works through WiFi? I was planning on reporting the RPMs back to the host using ESP-now on my case and having another ESP32 inside the computer and then writing the integration with FanControl, which I also use.

Another question, what's the amperage on the headers? Have you tested it with any watercooling pump? I need only 2 headers, one for 9 fans (about 1A total) and another one for the pump (max 2A at 100%)

SasaKaranovic, 2025-01-18:
Hey @LoLo2207 

You could use this board wirelessly but not on it's own. For example you could hook it up to Raspberry Pi (any would work).
BTW, the board uses standard USB connector for communication, so even if your setup is outside of your PC case, you could just run a USB cable back to your PC and it should work. For power you can get one of external SATA power adapters.

> Another question, what's the amperage on the headers? Have you tested it with any watercooling pump? I need only 2 headers, one for 9 fans (about 1A total) and another one for the pump (max 2A at 100%)

This question is not very easy to answer. There are many limiting factors, like your power supply, cables it uses, SATA connector, peak vs average current use etc. With that said, your maximum power draw will be limited by the "weakest link" in this chain.
The traces on the PCB should be able to easily handle 3A that you mentioned, but make sure everything else in the chain can do the same.
I have _not_ tested the board using any watercooling pumps (I don't have access to any). And honestly I think it might not work well with them, mainly because waterpumps and fans behaviour (RPM, TACH etc) will be much different than fan.

> BTW This project would be very interesting to the watercooling (and the MORA radiator) folks in reddit

Thank you! Great suggestion but I personally don't like when people shamelessly promote their projects/products or do it using puppet accounts, so because of this I won't post about it. But feel free to mention it or share your experience if you like.
My thoughts are that if something is good and worth mentioning, over time people will learn about it and share it with each other.

ThomasEg, 2025-01-22:
> Hey [@LoLo2207](https://github.com/LoLo2207)
>
> You could use this board wirelessly but not on it's own. For example you could hook it up to Raspberry Pi (any would work). BTW, the board uses standard [...]

I was actually looking for this exact information aswell. I run a "whole rack watercooling system" and have some industrial Noctua-fans running outside the house. These run at 0,5A each at full tilt. I was wondering what the PCB traces was rated at for this reason. I wouldn't want to let out the magic smoke ;-) I think most waterpumps also run at max 0,5A from the fan-connector, if higher power is required they usually come with a SATA power connector aswell. It could be interesting to do a load-test to the PCB and check if there is any hotspots...got a thermal camera anyone? :-)

> Thank you! Great suggestion but I personally don't like when people shamelessly promote their projects/products or do it using puppet accounts, so because of this I won't post about it. But feel [...]

I'm going to power one(maybe two) pumps through this board, hopefully. The pumps actually act quite like a fan, they accept PWM input and return actual RPM. How this RPM is then translated to to actual waterflow is another rabbithole not suiteable for this project ;-) But it is one of the things i hope to use the i2c port for...temperature and/or flow-sensor... Could be a series of add-ons for this board maybe? I would have ordered these from your shop if they were there... for now i just ordered one controller and see if i can make something work :-)

SasaKaranovic, 2025-01-23:
> I was actually looking for this exact information aswell. I run a "whole rack watercooling system" and have some industrial Noctua-fans running outside the house. These run at 0,5A each at full [...]

From the design perspective, traces are 2.5mm thick and that should easily carry at least 4-5A.
This is for per fan row, so 4-5A for top 5 fans (1-5), and another 4-5A for 5 bottom fans (6-10).
Obviously I did **not** test this, it's just numbers that I get from the design and back of the napkin calculation.

> I'm going to power one(maybe two) pumps through this board, hopefully. The pumps actually act quite like a fan, they accept PWM input and return actual RPM. How this RPM is then translated to to [...]

I do not have a water cooled system, so I can't test this. Even fans are not all standardized when it comes to RPM calculation/indication, but most commercial ones would produce 2 impulses per revolution. I would expect pump and fan to have a different RPM response curve, so the PID algorithm that is used for achieving target RPM on a fan, might not be compatible with the water pump.

But with all that said... there is only one way to find out. :)

closed 2025-02-13 by github-actions[bot]