SasaKaranovic/OpenFanController #32: Any models of the Latest Revision Hardware?

Senorgif2, 2025-03-05:
I received my openfan a few days ago, and I was planning on making a mount for it, but I cant seem to find any documentation or models of the latest PCB revision (June 2024).

ThomasEg, 2025-03-09:
Yeah, i've encountered the same issue. The enclosure here: https://github.com/SasaKaranovic/OpenFanController/blob/master/Enclosure/Bottom-PCB-Enclosure_by_Sam_Chan.3mf seems to be the new design, while the .step file and the Fusion360-file in the same folder seems to be an older design(3 holes vs. 2 holes for mounting). Maybe you can use the enclosure from Sam Chan to get started?

SasaKaranovic, 2025-03-11:
I have updated the [enclosure](https://github.com/SasaKaranovic/OpenFanController/tree/master/Enclosure) folder to include .step file of the board.
Also added another enclosure designed by Joe Dutson.

p.s. I don't know what kind of enclosure you are building and how precise/accurate your machines are, but I would say that for most (OpenFAN enclosure) use-cases, measuring from mesh or using calipers is more than accurate enough. Especially if you are 3D printing or laser-cutting an enclosure. :)

closed 2025-03-11 by SasaKaranovic

Senorgif2, 2025-03-11:
> I have updated the [enclosure](https://github.com/SasaKaranovic/OpenFanController/tree/master/Enclosure) folder to include .step file of the board. Also added another enclosure designed by Joe [...]

Thank you very much! No doubt I would have been able to figure it out, I have done something like that before, but it also usually means printing a few failed prototypes first. I am waiting for a lot of other parts for this project so I wasn't in a huge hurry, and I'd rather do it the easy way haha

SasaKaranovic, 2025-03-12:
No argument there. If you have the CAD model it's obviously much easier to do the measurements and work around it.
But I was also joking that we are getting a bit "spoiled" by having so much access to raw CAD/data, that if one piece is missing, we start flipping tables and rebelling. 😂 

Anyway, jokes aside, feel free to let me know if you have any issues with the model or need any additional support.