SasaKaranovic/OpenFanController #36: CoolerControl on Ubuntu

defaultsecurity, 2025-05-04:
Is it possible to use this controller with CoolerControl on Ubuntu?
https://gitlab.com/coolercontrol/coolercontrol

Coolercontrol supports devices via hwmon and liquidctl.

SasaKaranovic, 2025-05-15:
@defaultsecurity I'm sure it's possible because the controller is SW/HW agnostic. 😃 

I am more than happy to support anyone who wants to build any plugin/integration.
Feel free to reach out. There is also documentation available on https://docs.sasakaranovic.com/openfan/

SasaKaranovic, 2025-05-17:
> If I understood correctly, it's just using a standard serial port for communication, right? Simple but effective!

Correct, it uses serial port for communication with the hardware.
There is also a layer of abstraction (Web API) that allows easier integration and also allows multiples processes/apps to communicate with one controller locally and remotely. 

So if you don't want to lose the Web API functionality, it's probably best to integrate it by making http request.
Big downside to this approach is that you need to have the Web API app running in the background.

If you don't care about Web API (or maybe you will provide your own interface that other apps can use), then you should go directly to serial communication. This would be easier, more efficient and has no dependencies. But obviously you will lose (or have to implement your own) capability to allow other apps to communicate with the controller.

closed 2025-06-08 by github-actions[bot]