SasaKaranovic/OpenFanController #39: AttributeError: 'bool' object has no attribute 'split' on Raspberry Pi ARM

AlphaSoftwareInc, 2025-08-29:
Hello,

I have installed the Software via Docker on two different machines (Debian VM running on x64 and Raspberry Pi OS on Raspberry Pi 5 aarch64). On Debian it's running just fine but on Raspberry I get no RPM Speeds. Via the log I get this error message:

```
Aug 29 2025 18:59:47.296 pid:7 base_logger.py            59 INFO  Logging level: INFO
Aug 29 2025 18:59:47.305 pid:7 config.py                 49 DEBUG Server config:
Aug 29 2025 18:59:47.305 pid:7 config.py                 58 DEBUG {'hostname': 'localhost', 'port': 3000, 'communication_timeout': 1}
Aug 29 2025 18:59:47.305 pid:7 config.py                 60 DEBUG Hardware config:
Aug 29 2025 18:59:47.305 pid:7 config.py                 69 DEBUG {'port': None}
Aug 29 2025 18:59:47.305 pid:7 config.py                 71 DEBUG Fan profile config:
Aug 29 2025 18:59:47.305 pid:7 config.py                 80 DEBUG {'1000 RPM': {'type': 'rpm', 'values': [1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000, 1000], 'name': '1000 RPM'}, 'MadMax': {'type': 'PWM', 'values': [100, 100, 100, 100, 100, 100, 100, 100, 100, 100], 'name': 'MadMax'}, '1200RPM': {'type': 'RPM', 'values': [1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200, 1200], 'name': '1200RPM'}}
Aug 29 2025 18:59:47.305 pid:7 config.py                 83 DEBUG Fan aliases:
Aug 29 2025 18:59:47.305 pid:7 config.py                 92 DEBUG {0: 'Fan #1', 1: 'Fan #2', 2: 'Fan #3', 3: 'Fan #4', 4: 'Fan #5', 5: 'Fan #6', 6: 'Fan #7', 7: 'Fan #8', 8: 'Fan #9', 9: 'Fan #10'}
Aug 29 2025 18:59:47.305 pid:7 webserver.py             295 INFO  Using COM port `/dev/ttyACM0` (specified in `OPENFANCOMPORT` env variable).
Aug 29 2025 18:59:47.305 pid:7 webserver.py             304 INFO  Fan Controller port: /dev/ttyACM0
Aug 29 2025 18:59:47.307 pid:7 serial_driver.py          87 DEBUG Searching for COM port `/dev/ttyACM0`
Aug 29 2025 18:59:47.307 pid:7 serial_driver.py          89 DEBUG /dev/ttyACM0
Aug 29 2025 18:59:47.307 pid:7 serial_driver.py          90 DEBUG       Product: OpenFan
Aug 29 2025 18:59:47.307 pid:7 serial_driver.py          91 DEBUG       Desc: OpenFan - Board CDC
Aug 29 2025 18:59:47.307 pid:7 serial_driver.py          92 DEBUG       SN: DE6424128F7E2234
Aug 29 2025 18:59:47.307 pid:7 serial_driver.py          93 DEBUG       VID:11914 PID:10
Aug 29 2025 18:59:47.307 pid:7 serial_driver.py          94 DEBUG       Location: 3-1:1.0
Aug 29 2025 18:59:47.307 pid:7 serial_driver.py          95 DEBUG       Interface: Board CDC
Aug 29 2025 18:59:47.307 pid:7 serial_driver.py          97 DEBUG Using '/dev/ttyACM0' COM port
Aug 29 2025 18:59:47.309 pid:7 serial_driver.py          43 DEBUG Serial driver initialized with
Aug 29 2025 18:59:47.309 pid:7 serial_driver.py          44 DEBUG -- flush_on_write: 'True'
Aug 29 2025 18:59:47.309 pid:7 serial_driver.py          45 DEBUG -- timeout: '2'
Aug 29 2025 18:59:47.309 pid:7 webserver.py             334 INFO  Karanovic Research OpenFan - Starting API server
Aug 29 2025 18:59:47.310 pid:7 webserver.py             339 INFO  Listening on port 3000
Aug 29 2025 19:00:33.331 pid:7 webserver.py             257 DEBUG Requesting: None
Aug 29 2025 19:00:33.525 pid:7 webserver.py              53 INFO  Request:FANPROFILE_LIST
Aug 29 2025 19:00:33.532 pid:7 webserver.py             199 INFO  Request:GET ALL FAN Aliases
Aug 29 2025 19:00:34.544 pid:7 webserver.py             145 INFO  Request:STATUS
Aug 29 2025 19:00:34.544 pid:7 FanCommander.py           24 DEBUG CMD:0 Data: None
Aug 29 2025 19:00:34.544 pid:7 FanCommander.py           25 DEBUG Sending `>00`
Aug 29 2025 19:00:41.050 pid:7 serial_driver.py         145 ERROR Timeout occured (1756494041.0506754 > 1756494040.0444684)
Uncaught exception GET /api/v0/fan/status (192.168.1.60)
HTTPServerRequest(protocol='http', host='192.168.1.10:9000', method='GET', uri='/api/v0/fan/status', version='HTTP/1.1', remote_ip='192.168.1.60')
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/tornado/web.py", line 1846, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/mnt/OpenFan/webserver.py", line 147, in get
    rpm = self.handler.get_all_fan_rpm()
  File "/mnt/OpenFan/FanCommander.py", line 57, in get_all_fan_rpm
    return self._parse_fan_rpm(res)
  File "/mnt/OpenFan/FanCommander.py", line 38, in _parse_fan_rpm
    res = response.split("|")
AttributeError: 'bool' object has no attribute 'split'
500 GET /api/v0/fan/status (192.168.1.60) 6508.55ms
Aug 29 2025 19:00:41.547 pid:7 webserver.py             145 INFO  Request:STATUS
Aug 29 2025 19:00:41.547 pid:7 FanCommander.py           24 DEBUG CMD:0 Data: None
Aug 29 2025 19:00:41.548 pid:7 FanCommander.py           25 DEBUG Sending `>00`
```

Has anyone ever encountered this issue and/or knows how to fix this?

SasaKaranovic, 2025-09-05:
You have a line that indicates there was a timeout with the serial communication
`Aug 29 2025 19:00:41.050 pid:7 serial_driver.py         145 ERROR Timeout occured (1756494041.0506754 > 1756494040.0444684)`

But you are right, the software should more gracefully handle this edge case where board does not respond and a timeout occurs.

AlphaSoftwareInc, 2025-09-05:
Update: I flashed the firmware with the one from GitHub but still timeouts... 

How could I debug this any further? Also, is it know to be incompatible with Raspberry Pis / ARM ?